espressif/svd #2: CMSIS-SVD Schema Non-Conformance

jessebraham, 2021-07-20:
Due to the presence of the `<vendor>`/`<vendorID>`/`<licenseText>` tags, it appears that this SVD is targeting version 1.1 of the schema rather than 1.0 as currently reported in the SVD.

I used [xmllint](https://linux.die.net/man/1/xmllint) to perform validation on the current SVD against the CMSIS-SVD v1.1 Schema. Incorrect ordering or omission of certain elements can unfortunately result in errors:

>  ./esp32c3/ESP32C3.xml:7: element licenseText: Schemas validity error : Element 'licenseText': This element is not expected. Expected is ( description ).

> ./esp32c3/ESP32C3.xml:25: element cpu: Schemas validity error : Element 'cpu': This element is not expected. Expected is one of ( size, access, protection, resetValue, resetMask, peripherals ).

I manually fixed the above errors by:

* adding `<description>ESP32C3</description>` below `<version>` but before `<licenseText>`
    * This cannot be empty, so I have included a placeholder value for the time being
* moving the `<cpu>` block below `<licenseText>` but above `<addressUnitBits>`

Once the above changes have been made, a number of new errors appear. These all seem to be naming related, so I'm not sure how much we need to worry about them. I have saved the log in [this gist](https://gist.github.com/jessebraham/642b051913a5fad50dd4377c01f6ce85) for reference.

jessebraham, 2021-07-21:
Just a quick update, I've done some work on the SVD generation tool today and managed to fix most of the errors reported by `xmllint`. I still have a handful of issues I'd like to address but hopefully those will get done tomorrow.

aykevl, 2021-07-24:
Yes, I think the naming warnings are in fact bugs:

```xml
    <peripheral>
      <name>SPI</name>
      <description>Peripheral SPI</description>
      <baseAddress>0x60024000</baseAddress>
      <size>32</size>
      <registers>
        <cluster>
          <name>User-defined control registers</name>
          <description>User-defined control registers</description>
```

Here, it looks like the human-readable description was copied to the name tag. This is an issue for languages that want to use the name tag to generate struct field names or method names.

That said, the first two errors:

```
./esp32c3/ESP32C3.xml:24: element name: Schemas validity error : Element 'name': [facet 'enumeration'] The value 'RISC-V' is not an element of the set {'CM0', 'CM0PLUS', 'CM0+', 'CM1', 'SC000', 'CM23', 'CM3', 'CM33', 'CM35P', 'SC300', 'CM4', 'CM7', 'ARMV8MML', 'ARMV8MBL', 'ARMV81MML', 'CA5', 'CA7', 'CA8', 'CA9', 'CA15', 'CA17', 'CA53', 'CA57', 'CA72', 'other'}.
./esp32c3/ESP32C3.xml:25: element revision: Schemas validity error : Element 'revision': [facet 'pattern'] The value '1' is not accepted by the pattern 'r[0-9]*p[0-9]*'.
```

Are totally expected. This is not a Cortex-M family CPU, so these are expected to fail. That said, it might be useful to put something more relevant in the CPU name field, such as `RV32IMC` or even the actual name of the CPU (if it isn't designed by Espressif).

Finally, the description of each peripheral is just `Peripheral <name>`, which is perhaps not very useful (although not a big issue either).

aykevl, 2021-07-24:
With some further investigation I found that the way clusters are used does not follow the intended use of them. Quoting from the spec:

> It is predominantely targeted at the generation of device header files to create a C-data structure within the peripheral structure instead of a flat list of registers.

In the case of this SVD file, clusters appear to be used to group registers that are related in function but not in location in memory. In other words, the various clusters overlap. This makes it impossible to create C-like structs from these clusters.
It can be worked around by ignoring the clusters and flattening all registers, but that doesn't seem like a great way to deal with this issue. (That would as a side effect also avoid issues with the non-conforming names).

jessebraham, 2021-07-26:
Thank you for your input @aykevl!

Regarding names, there were indeed a few errors made and I believe they have mostly been corrected now. These changes have not yet been made public but I will try to get an updated version of the SVD pushed to this repo soon.

As for the peripheral descriptions, the source files unfortunately do not include this information so we are just using placeholder values for the time being. This may change at some point in the future but it is not presently a high priority.

We will discuss the cluster issues internally and come up with a solution for that, thank you for bringing it to my attention.

jessebraham, 2021-08-04:
I believe all of the points discussed here have been fixed in #4.

closed 2021-08-04 by jessebraham