espressif/svd #6: <interrupt> sections in peripherals missing

MabezDev, 2021-08-13:
Without the relevant `<interrupt>` sections within the `<peripheral>` sections we will have to generate and patch them in for each like we do for the [janky esp32 crate](https://github.com/esp-rs/esp32/blob/master/svd/patches/_interrupts.yaml).

Is it possible to add the relevant info from the internal format?

igrr, 2021-08-13:
Which interrupt numbers would you like to have available in the SVD file — the interrupt matrix inputs for each peripheral, or the CPU interrupt inputs? From interrupts.yml you linked, it seems that these are the interrupt matrix inputs. If that is the case, where should we store the information about CPU interrupt inputs: their interrupt types (edge/level/timer/software/profiling/nmi) and priority levels?

MabezDev, 2021-08-13:
> where should we store the information about CPU interrupt inputs: their interrupt types (edge/level/timer/software/profiling/nmi) and priority levels?

That's a good question! For the esp32, [we hacked them in a peripheral interrupt sources too](https://github.com/esp-rs/esp32/blob/b66b063e71e5ea1d2223a5fdc569e0dde35dfb8c/svd/patches/_interrupts.yaml#L257-L280). Then we had a [special case in our interrupt handling in the hal](https://github.com/esp-rs/esp32-hal/blob/master/src/interrupt.rs#L187-L197) to always map those Interrupts to their respective  CPU interrupt.

I don't really think that's a good idea, perhaps we should expose them in the `xtensa-lx-rt` crate instead?

igrr, 2021-08-13:
> I don't really think that's a good idea, perhaps we should expose them in the xtensa-lx-rt crate instead?

That's an option! The (minor) issue is that the CPU interrupt list is actually configuration-specific, so some other Xtensa LX based chip might have a totally different list compared to the ESP32. Lucky for us, ESP32, ESP32-S2 and ESP32-S3 all use the same configuration, so we can pretend, for now, that it's indeed something common for all Xtensa LX chips. If some vendor (or ourselves!) comes up with a chip that has a different list of CPU interrupts, we can discuss this issue again.

MabezDev, 2021-08-13:
Yes I think the xtensa-lx* crates could do with a bit of an overhaul in that regard. They still work on the assumption that `lx$NUMBER` has a set configuration.

https://github.com/esp-rs/xtensa-lx/blob/1da638a9113182be0f90eb7f1104b9245e0daa7d/Cargo.toml#L20-L33

These crates were created when I knew a lot less about the Xtensa arch, so I think we should replace the `lx6` & `lx106` with `esp32` and `ep8266` respectively. Then if we or another vendor wants supports new chip, they can simply PR thier chip via a feature flag i.e:

```
[features]
# our new awesome chip from AwesomeChipCo was only configured with ccount support
newAwesomeChip = ["ccount"]
```

jessebraham, 2021-10-06:
Just a small update.

Some peripherals now have their interrupts listed in the SVD. Hopefully with the next release we can get the rest in there, but there are still a handful of missing peripherals which must be dealt with first.

jessebraham, 2021-11-26:
Each SVD has the interrupts present for whichever peripherals are covered, so I'm going to consider this issue fixed. I opened an issue regarding the CPU interrupts here: https://github.com/esp-rs/xtensa-lx-rt/issues/29

closed 2021-11-26 by jessebraham