espressif/svd #7: Merging large array based registers

MabezDev, 2021-08-13:
Whilst most of the patches in `esp32` are hacks to supplement the broken svd, some are actually useful QoL improvements.

- [gpio.yaml](https://github.com/esp-rs/esp32/blob/master/svd/patches/_gpio.yaml) patch merges `FUNC*_IN_SEL_CFG` & `FUNC*_OUT_SEL_CFG`.
- Not valid for esp32c3 (I think) but [rtcio](https://github.com/esp-rs/esp32/blob/master/svd/patches/_rtcio.yaml) merges `PIN*` registers.

These patches reduce the size of the svd & allow `svd2rust` to generate code that can access these registers via index, instead of by name.

jessebraham, 2021-08-13:
This would definitely be an improvement over the current version, I agree. This was brought up in #3 as well and I am actively investigating it!

jessebraham, 2021-08-30:
The noted registers have been collected into arrays in 11b28a1.

closed 2021-08-30 by jessebraham