espressif/svd #9: UART-FIFO R/W field classified as read-only

ducktec, 2021-08-23:
The new SVD file with the register access information (commit 9f73559) specifies the field `UART`->`FIFO`->`RXFIFO_RD_BYTE` as `read-only`:
```xml
<field>
  <name>RXFIFO_RD_BYTE</name>
  <description>UART 0 accesses FIFO via this register.</description>
  <bitOffset>0</bitOffset>
  <bitWidth>8</bitWidth>
  <access>read-only</access>
</field>
```
The ESP32-C3 TRM v0.3 states in section 20.4.2:
> UARTn can access FIFO via register UART_FIFO_REG. You can put data into TX FIFO by writing
> UART_RXFIFO_RD_BYTE, and get data in RX FIFO by reading UART_RXFIFO_RD_BYTE.

but the register description in the TRM (section 20.6 and 20.7) describes the field and register as read-only.

It was possible to send data by writing to the field, so I would assume that it should be marked as `read-write`.

I suspect (by comparing the registers with the S2 TRM) that the same is also applicable to `I2C`->`FIFO_DATA`->`FIFO_RDATA`, but as the I2C peripheral is not yet documented in the C3 TRM v0.3, this is just a suspicion.

_I'm not sure if this is the right way to report these kind of issues. I can also report them via the documentation feedback form if that is preferred._

jessebraham, 2021-08-23:
Thank you for opening this issue, this repo is indeed the correct place for these!

Long story short, our SVD is being generated from a set of files in an internal-only format from our digital team. There seem to be some errors in these files, so I suspect that is where these are originating. We have just recently begun working on SVD generation so we're still working out some kinks, and it will likely take some time to get these types of issues ironed out.

I will be sure to update this issue once I have any further updates.

jessebraham, 2021-08-30:
This has been resolved in 11b28a1, thank you again for opening the issue!

closed 2021-08-30 by jessebraham