espressif/svd #13: Some registers have the incorrect size

jessebraham, 2021-09-28:
The SVD generator at present assumes that all registers are a fixed width of 32 bits. This is not a correct assumption, however.

For example, in the `XTS_AES` peripheral, the `PLAIN_MEM` register should have a width of 128 instead.

aykevl, 2021-09-30:
> For example, in the `XTS_AES` peripheral, the `PLAIN_MEM` register should have a width of 128 instead.

Hmm, how should programs use this? There are no 128-bit instructions. And while 64-bit types are common in programming languages, not all programming languages have 128-bit types (Go does not, and C only sometimes).
However, this is perhaps best left to SVD processing tools. They can decide what to do with a 128 bit type instead (for example, in Go it could be converted to `[4]uint32` or `[16]byte`).

jessebraham, 2021-10-06:
I'm not entirely sure how to handle this, I've been trying to look into potential solutions. I was hoping that `svd2rust` (which is my primary target) had some sort of slice API, but I was not so lucky.

Really these memory regions should not be classified at registers at all, however as far as I can tell the CMSIS-SVD specification does not offer any solution for this problem.

aykevl, 2021-10-06:
What about this?

```xml
        <register>
          <dim>16</dim>
          <dimIncrement>0x01</dimIncrement>
          <name>PLAIN_MEM[%s]</name>
          <description>...</description>
          <addressOffset>...</addressOffset>
        </register>
```