espressif/svd #19: I2C SVD Observations

ducktec, 2022-02-20:
I started implementing a shared I2C implementation for all ESP32 variants for `esp-hal` and noticed a few issues with the SVDs:
- In the ESP32-S2 SVD, the `COMD%s` registers are not combined in an array as with the other.
- For the ESP32, S2 and S3 variants, the I2C FIFO register is called `DATA` in the SVD, only for the C3 it’s called `FIFO_DATA`. However, in the data sheets it’s the same (`DATA`) with all variants.
- For the ESP32, the DPORT registers `PERIP_CLK_EN` and `PERIP_RST_EN` have only one 32-bit field. According to the datasheet they should have dedicated bit-sized fields for the individual peripherals. Otherwise it's very difficult to enable/disable individual peripherals.
- In the ESP32 SVD a `DATA` register at offset `0xC0` is defined for the I2C peripheral. However, according to the ESP32 TRM, no such register exists. Is this register really existing and can be relied upon, or is this a SVD error?
- For the S2, S3 and C3 variants, the field `I2C_FIFO_RDATA` in `I2C_DATA_REG` is marked as _read-only_ which matches the TRM register description, but the TRM wording seems to indicate for all three that that field is _read-write_ (S2 TRM section 25.3.2.1, C3 TRM section 25.4.10, S3 TRM section 23.4.10): 
  > In FIFO mode, the CPU writes to TX RAM via fixed address I2C_DATA_REG, with addresses for writing in TX RAM incremented automatically by hardware.
  > [...]
  > In FIFO mode, the CPU reads RX RAM via fixed address I2C_DATA_REG, with addresses for reading RX RAM incremented automatically by hardware.

  At least for the C3 variant I can confirm that writing to that field works and yields in the expected behavior.

jessebraham, 2022-02-22:
Thanks for reporting this issue! I'm currently in the process of fixing a number of problems (now including these ones), so at least some of these should be fixed with the next version of the SVDs.

I have addressed the first two points you mentioned. I will update this issue as I come to your other points.

jessebraham, 2022-02-23:
I believe all of the points raised in this issue have been addressed with the most recent releases. If this is not the case please feel free to re-open this issue, or to open a new one.

closed 2022-02-23 by jessebraham