espressif/svd #22: Add Releases for SVD files ?

brianignacio5, 2022-07-14:
Could we add resulting SVD as release files ? In this way we could download them from any scripting, IDE, command line tool that wants to work with them.

Also enabling mirrors in Espressif DL servers.

aykevl, 2022-07-15:
One place where they are stored is here: https://github.com/posborne/cmsis-svd/tree/master/data/Espressif
Probably out of date, but it might be useful.

jessebraham, 2022-07-15:
Not against doing this but I'm not really sure how to go about accomplishing it. SVDs are updated irregularly (they do not get updated all at once) and the version numbers are not in sync.

igrr, 2022-07-15:
How about making an automated release whenever something gets updated, using the current YYYYMMDD date as the release tag? Not ideal, but at least it's going to be pretty easy to set up and won't require changes to the project workflow.

igrr, 2022-07-15:
We could also maintain the "latest" release using https://github.com/eine/tip.

PR #23 (2022-08-25): Add a CI workflow to create a release whenever `main` is pushed to
Whenever the `main` branch is pushed to, this will action will create a tag and a release, both named using the current date (eg. `2022-07-22`). Each release will contain all SVD files, regardless of whether or not a specific file has been updated. An example release can be found [here](https://github.com/jessebraham/svd/releases/tag/2022-07-22).

@igrr @brianignacio5 does this seem suitable?

Closes #22.
closed 2022-08-25 by jessebraham