espressif/svd #24: ESP32-C3 TWAI Data Register Access

alexbohm, 2022-09-08:
While implementing a Rust driver, I came across some inconsistencies in the access to the TWAI_DATA_x_REG registers. It seems like they are listed as write only access in the svd, yet I believe they are used to read from the receive buffer.

Snippet of `esp32c3.svd`:
```xml
        <register>
          <name>DATA_0</name>
          <description>Data register 0</description>
          <addressOffset>0x40</addressOffset>
          <size>0x20</size>
          <fields>
            <field>
              <name>TX_BYTE_0</name>
              <description>In reset mode, it is acceptance code register 0 with R/W Permission. In operation mode, it stores the 0th byte information of the data to be transmitted under operating mode.</description>
              <bitOffset>0</bitOffset>
              <bitWidth>8</bitWidth>
              <access>write-only</access>
            </field>
          </fields>
        </register>
```

If I'm reading section 29.4.4.1 in the ESP32-C3 reference manual correctly, it states that writing these addresses is for transmission and reading accesses the receive buffer:
https://www.espressif.com/sites/default/files/documentation/esp32-c3_technical_reference_manual_en.pdf#subsubsection.29.4.4

To add to the confusion, it looks like reading these registers is undocumented in the ESP32-C3 reference manual outside of that section.

Taking a look at how the idf C code implements the drivers:
https://github.com/espressif/esp-idf/blob/867745a05c5cb1fe5eb212200580626ad0fa839d/components/hal/esp32c3/include/hal/twai_ll.h#L514
it does look like these registers are read to receive packets.

alexbohm, 2022-09-09:
Thanks for taking a look at this! I just put up a pull request with a patch. I'm not too familiar with what the SVD files would look like for the two operating modes, but for now having the read-write works. Thanks!

jessebraham, 2022-09-09:
Unfortunately the SVD specification does not seem to support multiple modes. I had discussed this briefly with a colleague, the thoughts were that we can do one of the following (or possibly another option):

1. Create _two_ TWAI peripherals in the SVD, one for each mode
2. If able, just update the access level and/or whatever else needs updating, and the developer will need to be aware of some registers having multiple uses

Not sure which the best solution is, but it may become evident once the driver is implemented.