espressif/svd #25: ESP32 RNG Peripheral Mapping

fremdrost, 2022-09-17:
Maybe I am wrong, but should the base address of the RNG be 0x3FF75000 instead of 0x60035000?

Snippet of `esp32.svd`:
>
    <peripheral>
      <name>RNG</name>
      <description>Hardware random number generator</description>
      <groupName>RNG</groupName>
      <baseAddress>0x60035000</baseAddress>
      <addressBlock>
        <offset>0x0</offset>
        <size>0x4</size>
        <usage>registers</usage>
      </addressBlock>
      <registers>
        <register>
          <name>DATA</name>
          <description>Random number data</description>
          <addressOffset>0x144</addressOffset>
          <size>0x20</size>
        </register>
      </registers>
    </peripheral>

[https://espressif.com/sites/default/files/documentation/esp32_datasheet_en.pdf](https://espressif.com/sites/default/files/documentation/esp32_datasheet_en.pdf)

jessebraham, 2022-09-20:
The peripheral can be accessed using the data or instruction bus, which is why the addresses are different. It is probably more correct to use the other address you've specified, so I'll update that whenever I get around to it. In the meantime the peripheral still seems to be working in `esp-hal`, so the SVD should be fine.

fremdrost, 2022-09-24:
Oh, you're right. I missed that.

Thanks.

closed 2023-03-10 by jessebraham