espressif/svd #26: esp32c3: RNG_DATA_REG should have read-only access

lucab, 2023-01-27:
The `RNG_DATA_REG` definition `esp32c3.svd` seems to be partially off:
```xml
    <peripheral>
      <name>RNG</name>
      <description>Hardware random number generator</description>
      <groupName>RNG</groupName>
      <baseAddress>0x60026000</baseAddress>
      <addressBlock>
        <offset>0x0</offset>
        <size>0x4</size>
        <usage>registers</usage>
      </addressBlock>
      <registers>
        <register>
          <name>DATA</name>
          <description>Random number data</description>
          <addressOffset>0xB0</addressOffset>
          <size>0x20</size>
        </register>
      </registers>
    </peripheral>
```

The register lacks a `<access>read-only</access>` to match the description in section 24.5 of the [manual](https://www.espressif.com/sites/default/files/documentation/esp32-c3_technical_reference_manual_en.pdf).

I'm not sure what's the correct semantic default when the `access` property is missing from a register, but at least for the Rust bindings this seems to translate to `read-write` access: https://docs.rs/esp32c3/0.10.0/esp32c3/rng/data/index.html

jessebraham, 2023-01-27:
Thanks for opening this issue.

I've made this correction for all devices in our SVD generator, so this should be fixed with the next round of SVDs. I will leave this issue open until those have been published, I don't have an estimated time frame at this point.

lucab, 2023-01-27:
Thanks for the very prompt reaction! :heart:

closed 2023-03-10 by jessebraham