espressif/svd #35: Wi-Fi hardware register documentation

redfast00, 2024-09-18:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

We've been working for close to a year on the [esp32-open-mac project](https://github.com/esp32-open-mac/esp32-open-mac/) where we're building an open source Wi-Fi MAC layer for the ESP32. This has several advantages, for example being able to implement 802.11s compliant mesh networking, better auditability, ... A lot of the effort involved was around reverse engineering the hardware registers relevant to Wi-Fi since these are all undocumented.

We currently have a proof of concept that (after still using the proprietary radio initialisation/calibration) is able to connect to an open Wi-Fi network using only open source code. We used ESP-NETIF to connect our Wi-Fi driver to LWIP, so we can then ping the ESP32 or even make HTTP requests. We spent a lot of time both building tooling to reverse engineer and actually reverse engineering the hardware, and even then it took some trial and error to figure out the functionality of the registers we're currently using.

One thing that would really help us, is any and all documentation that can be released. I understand that there are certain parts that might need to be kept private, but any documentation is helpful. Right now, we'd really be helped with documentation related to the Wi-Fi MAC peripheral (for example, information about the MAC RX filters, WPA hardware acceleration, ...).

/CC @Spritetm @igrr

chrysn, 2024-09-23:
Any support to the esp32-open-mac would not only benefit that project itself, but also other projects that can then use it. For example, the runtime use of the ESP32 SDK in RIOT OS has caused problems over and over, and esp32-open-mac would allow a much smoother integration.

Please give any support you can; register names should be easy to do even when there are legal reasons why the SDK can not be fully disclosed.

Frostie314159, 2024-09-23:
The thing is, that even just register names would help us a lot. We've been building our own understanding of how the hardware works through RE. For some registers, the meanings are pretty obvious and can be easily deduced, but some registers only have one or two references, which don't give us any information at all. 

For IEEE 802.15.4 the registers are in the SVD file and there is some public code using them, but I guess the IP situation is a different one there.