espressif/svd #36: ESP32C5 SVD?

brainstorm, 2025-05-23:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Described the feature in detail and justified the reason for the request.
- [x] Provided specific use cases and examples.

### Feature description

Hello!

I have a brand new ESP32C5 devkit sitting on my desk and I've just noticed that there's no public SVD released yet for it?.. without it it's not easy to [generate a Rust PAC for it](https://github.com/esp-rs/esp-hal/issues/3454).

### Use cases

My target/specific use case is ssh-stamp supporting all wifi-enabled targets (see: https://github.com/brainstorm/ssh-stamp/pull/40)... not a critical pain point at this point in time though, just curious.

### Alternatives

I will NOT go through the TRM PDF and generate a SVD for it: https://github.com/brainstorm/datasheet2svd ... nope, not this time ;)

igrr, 2025-05-23:
Specifically for your Rust use case: esp-pacs actually uses different SVD files (modified compared to the ones in this repo). For that, please subscribe to https://github.com/esp-rs/esp-hal/issues/3454 if you haven't already.

As for this repo, we will add the C5 SVD, request noted!

brainstorm, 2025-05-23:
Thanks @igrr! Yes, I'm aware that [`.yaml` patches are needed for svd2rust & co](https://github.com/esp-rs/esp-pacs/blob/main/esp32c6/svd/patches/esp32c6.yaml), but IIRC the first step is getting the "base" SVD to patch from, looking forward to have that one available 👍🏻

Jason2866, 2026-03-06:
+1 for missing C5 svd file

Terstegge, 2026-09-22:
Also waiting for this file... Any progress yet?